adafruit/Adafruit_LSM9DS1 #1: No need to wait after Wire.requestFrom() and it should not be followed by a Wire.endTransmission()

Koepel, 2017-02-03:
In Adafruit_LSM9DS1.cpp in the function readBuffer() in I2C mode, there is a while-loop after the Wire.requestFrom(). However, there is no such thing to wait for data. The Wire.requestFrom() returns when the I2C transaction has completely finished and the received data is in a buffer in the Wire library. In case of a bus collision in a multi-master bus or a bad hardware I2C bus, the Wire.available() could return zero bytes, and that while-loop halts the sketch. Therefor, that while-loop has no function at all and in a (very rare) situation it can do harm.

The Wire.requestFrom() should not be followed by Wire.endTransmission(). The Wire.endTransmission() is only used when sending data. When it is used after Wire.requestFrom() in the Wire library for AVR chips, an extra I2C bus transaction takes place: the address is sent onto the bus and the sensor replies with acknowledge. This has no further consequences with the Wire library for AVR chips, but it is not needed, and it is wrong use of the Wire library, and it slows the code down with an extra I2C transaction. When used with other Arduino boards or software implementations of the Wire library, the result could be different.

As you may have noticed, I have written a lot of similar issues. I started to do this because I had to explain in Arduino forums too many times that the Wire library was used wrong. Since Adafruit is ahead of others, your code is often used as an example how to write code for Arduino. With this very long issue, I really hope you stop using the Wire library in the wrong way.

ladyada, 2017-02-03:
https://github.com/adafruit/Adafruit_LSM9DS1/commit/62af8112ade84012bc5165d0d4693591ce1bed15

?

closed 2017-02-03 by ladyada

Koepel, 2017-02-03:
Yes, thank you.
There is still a Wire.endTransmission() after the Wire.requestFrom(). It is at the moment at line 443.

Koepel, 2017-02-10:
I noticed that the unnecessary Wire.endTransmission() also has been removed. Thank you. Now this issue is really closed.