adafruit/Adafruit_LSM9DS1 #5: Working with PJRC MotionCal?

wormyrocks, 2017-12-14:
I'm trying to use this library to implement a Mahony filter; I'd like to use MotionCal to determine calibration constants for the magnetometer.

I added methods to expose the raw int16 values from the sensors - however, I get garbage data (or at least it renders nothing close to a sphere) when running MotionCal.

Has anyone else had luck with this? I found [one other thread](https://forum.pjrc.com/threads/43838-MotionCal-Tool-Format) about it on the PJRC forums, but it had no responses.
thanks!

Sindel, 2018-10-26:
Hey, did you manage to make it work? Let me know!

caternuson, 2018-10-26:
Closing this as it seems like an issue with an attempt to modify the library.

FWIW - You could try using the `readBuffer` function to get the raw values:
https://github.com/adafruit/Adafruit_LSM9DS1/blob/master/Adafruit_LSM9DS1.h#L199
See here for an example of how it's used to grab to accel values:
https://github.com/adafruit/Adafruit_LSM9DS1/blob/master/Adafruit_LSM9DS1.cpp#L170
Luckily it's public for this class, so you're free to call it from your user code.

closed 2018-10-26 by caternuson