adafruit/Adafruit_LSM9DS1 #15: Temperature doesn't seem to report correctly..?

ForbiddenEra, 2021-03-22:
Hi,

I am in the process of testing several IMUs [MPU9250*, Ada FXOS/FXAS breakout, Ada LSM9DS1 breakout, a BNO55 and a BNO80.

(* I believe the 9250 is EOL so, only testing because I have on hand but likely won't make it into the product)

I've got it running with the MPU9250 and the LSM9DS1 right now. The LSM9DS1 is reporting a weird temperature value. The value does increase if I warm the chip up with my finger, but it's not deg C. It was originally sitting at 21, now it's at 53 after going up to 149 after a few seconds with my finger on it, 117, 85.. it's jumping by 32 every time..?

Dividing by 32 maybe gives the delta temperature in C, but I'm not 100% sure if that's the case or that's just garbage.. After holding both chips for the same amount of time, the delta on the MPU was about 4.2 degrees, 181/32 is 5.65 degrees, it started at 53 = 3.99 deg C?

PS: Unified library is great, wish it supported more! Also wish I could use libraries like this on my radio modules but the python they have is so stripped down (snappy)

ladyada, 2021-03-22:
yep the lsm9ds1 temperature sensor is not 'absolute' its relative.

closed 2021-03-22 by ladyada

ForbiddenEra, 2021-03-22:
Ah, I probably should have checked the datasheet first ;-)

thanks..I've been albe to use your lib for all the IMU I'm testing except 9250, when I said I hoped for more, I didn't see the BNO55 and BNO8X listed on the page for it.. so that was a nice find!! using it flawlessly with BNO055, BNO080, LSM9SD1 and the FXOS/FXAS combo board..didn't see one for mp9250 tho :)

thanks again for your hard work